Heckie75/eQ-3-radiator-thermostat #1: `status` returns `can't read "d": no such variable`

gismo141, 2017-04-05:
Hallo @Heckie75,

Vielen Dank erstmal für die tolle Arbeit - habe dein Projekt über das FHEM-Forum gefunden und bin gerade dabei es auszuprobieren.

Leider bekomme ich das jedoch nicht hin.
Ich habe zuerst die Adresse von einem meiner Thermostate identifiziert (`00:1A:22:06:32:30`).
Anschließend die benötigte Software installiert (`TCL 8.4` und `expect`).

Muss ich das Thermostat vorher noch irgendwie anlernen?
Führe ich direkt eine `status`-Abfrage durch kommen folgende Fehlermeldungen:

```
$ ./eq3.exp 00:1A:22:06:32:30 status
can't read "d": no such variable
    while executing
"return $d"
    (procedure "hexToDecimal" line 3)
    invoked from within
"hexToDecimal [lindex $raw_status 3]"
    (procedure "parseMode" line 7)
    invoked from within
"parseMode $raw_status"
    (procedure "readMode" line 6)
    invoked from within
"readMode"
    (procedure "init" line 11)
    invoked from within
"init [expr $INIT(device) + $INIT(mode) + $INIT(timer)]"
    ("^status$" arm line 2)
    invoked from within
"switch -regexp $c {
    ^status$ {
      init [expr $INIT(device) + $INIT(mode) + $INIT(timer)]
      dumpThermostat
    }
    ^sync$ {
      init 0
 ..."
    (procedure "doCommand" line 10)
    invoked from within
"doCommand"
    (file "./eq3.exp" line 929)
```

Heckie75, 2017-04-05:
Hallo,

das ist ja schade. Ggf. ist es notwendig auch noch Tclx zu installieren:
```
sudo apt-get install tclx8.4
```

Weiterhin könntest Du testen, ob das Thermostat überhaupt antwortet. Folgende Befehle sollten den Status zurückliefern:
```
$ gatttool -b 00:1A:22:06:32:30 -I
[00:1A:22:06:32:30][LE]> connect
Attempting to connect to 00:1A:22:06:32:30
Connection successful
[00:1A:22:06:32:30][LE]> char-write-req 0411 03
Notification handle = 0x0421 value: 02 01 00 00 04 2a
```
siehe auch https://github.com/Heckie75/eQ-3-radiator-thermostat/blob/master/eq-3-radiator-thermostat-api.md

Das Kommando `char-write-req 0411` ist die zentrale Funktion. Falls es hier keinen `Notification handle` gibt, ist irgendwas noch nicht richtig. Vielleicht ist doch noch ein Pairing notwendig?

Vielleicht das noch mal probieren. Bin auf das Ergebnis gespannt. 

Viele Grüße

gismo141, 2017-04-05:
Vielen Dank für die schnelle Antwort!

Ja es scheitert bereits beim `Notification handle`.
Ich habe nun folgendes probiert:

```
$ bluetoothctl
[bluetooth]# untrust 00:1A:22:06:32:30
    [CHG] Device 00:1A:22:06:32:30 Trusted: no
    Changing 00:1A:22:06:32:30 untrust succeeded
[bluetooth]# disconnect 00:1A:22:06:32:30
    Attempting to disconnect from 00:1A:22:06:32:30
    Successful disconnected
[bluetooth]# remove 00:1A:22:06:32:30
    [DEL] Device 00:1A:22:06:32:30 CC-RT-BLE
    Device has been removed

$ gatttool -b 00:1A:22:06:32:30 -I
[00:1A:22:06:32:30][LE]> connect
    Attempting to connect to 00:1A:22:06:32:30
    Connection successful
[00:1A:22:06:32:30][LE]> char-write-req 0411 03
    Error: Characteristic Write Request failed: A timeout occured
```

Wenn ich anschließend das Thermostat neu anlerne passiert folgendes:
```
$ bluetoothctl
[bluetooth]# trust 00:1A:22:06:32:30
    Device 00:1A:22:06:32:30 not available
[bluetooth]# scan on
    Discovery started
    ...
    [NEW] Device 00:1A:22:06:32:30 CC-RT-BLE
    ...
[bluetooth]# scan off
    Discovery stopped
[bluetooth]# trust 00:1A:22:06:32:30
    [CHG] Device 00:1A:22:06:32:30 Trusted: yes
    Changing 00:1A:22:06:32:30 trust succeeded
[bluetooth]# pair 00:1A:22:06:32:30
    Attempting to pair with 00:1A:22:06:32:30
    [CHG] Device 00:1A:22:06:32:30 Connected: yes
    [CHG] Device 00:1A:22:06:32:30 Paired: yes
    Pairing successful
[bluetooth]# exit

$ gatttool -b 00:1A:22:06:32:30 -I
[00:1A:22:06:32:30][LE]> connect
    Attempting to connect to 00:1A:22:06:32:30
    Error: connect: Device or resource busy (16)
[00:1A:22:06:32:30][LE]> exit

$ bluetoothctl
[bluetooth]# disconnect 00:1A:22:06:32:30
    Attempting to disconnect from 00:1A:22:06:32:30
    Successful disconnected
    [CHG] Device 00:1A:22:06:32:30 Connected: no
[bluetooth]# exit

$ gatttool -b 00:1A:22:06:32:30 -I
[00:1A:22:06:32:30][LE]> connect
    Attempting to connect to 00:1A:22:06:32:30
    Error: connect error: Transport endpoint is not connected (107)
```

Sehr seltsam, oder mache ich da irgendwas eindeutig falsch?

Viele Grüße,
Michael

closed 2017-10-15 by Heckie75