Heckie75/eQ-3-radiator-thermostat #2: char-write-req 0411 03 corrupts memory state

danielfaust, 2017-09-13:
On my device, if I issue an `char-write-req 0411 03` without specifying the datetime like `char-write-req 0411 03YYMMDDHHMM` the internal state of the memory seems to get corrupted. Ie. if I then long-press menu and select `dat` I don't get 2017 but garbled stuff like 23E8 and I can't set the vacation mode via the wheel properly. Basically all time-related stuff is then messed up.

Heckie75, 2017-09-13:
Is it something that you have directly done in gatttool? Then I would warn in API description. 

Or is it possible to force this situation by using the CLI tool that I have written? Then I must change parameter validation. 

Is the system fine again after a factory reset?

Heckie75, 2017-09-13:
Does a factory reset help?

Do you think that you have a different firmware version?
What is the output of "characteristics"?

danielfaust, 2017-09-13:
When I write the datetime to the device all seems to go back to normal. I didn't do a device reset because I think that I then have to re-mount it on the heater so that it can calibrate itself before being able to communicate with it again, too lazy to do that. The output is mostly normal, except when in vacation mode then the last 4 bytes reflect what is corrupted in memory. Those are additional bytes that are only returned in vacation mode and are supposed to show the current vacation mode configuration.

Heckie75, 2017-09-13:
Hi Daniel,

Thanks for your investigation. At least I should update the API doc and of course also the CLI tool. I have understood that it is not appropriated to send this in order to get status:

  char-write-req 0411 03

It is always required to update the time as well in order to keep the vacation mode running and the system stable. Correct?

Best regards,
Martin

Heckie75, 2017-10-15:
I have changed the script so that time will be updated even if it just a status request - simular to official app on Android.

closed 2017-10-15 by Heckie75

vcmorini, 2019-08-22:
Hello, is there a correct order to update/sync? After every command or before?
Thanks :)

Heckie75, 2019-08-22:
Sync must be done once in order to set the internal clock.
There is no need to sync explicitly before wach command.

vcmorini, 2019-08-30:
> Sync must be done once in order to set the internal clock.
> There is no need to sync explicitly before each command.

thks Heckie75