Heckie75/eQ-3-radiator-thermostat #4: can't connect after some minutes

julianvollmer, 2017-10-26:
Hi @Heckie75,
thanks for your work, everything with your code works fine.
But my thermostat device doesn't advertise all the time.
When I remove the battery set date, time everything works fine for a unknown time (some minutes).
Put the device in scan mode seems to be useless.

Do you have any experience with a behavior like this ?

Cheers 
Julian

Heckie75, 2017-10-27:
Hi Julian,

I don't expect that the thermostat has something like a bluetooth standby mode. As long as bluetooth is turned on you should be able to re-connect at any time.
Of course, you are able to shutdown bluetooth of the thermostat at all, but that's probably not what you mean. Maybe you can double-check the bluetooth settings by pressing the "menu" button for 3 seconds and go to the bluetooth settting. Here I can toggle the setttings. If bluetooth is "on", I can confirm to set it "off" and vice versa.

Have you tried to connect to the device directly by using gatttool?

What's about the official app? Do you face the same problem here?

Best regards,
Martin

PS: According Amazon's customer reviews other user's report simular problems

julianvollmer, 2017-10-27:
Hi @Heckie75,

I've updated the thermostat with the Smartphone App to Version 1.10. It looks like it's advertising all the time and I can always connect to it.

So, at the moment everything looks pretty well.

Best regards,

Julian

closed 2017-10-28 by Heckie75