Heckie75/eQ-3-radiator-thermostat #5: Requires package Tclx

SelmaUrban, 2017-11-05:
Hi Heckie75, 
first i want to thank you for this great script!

I saw that you have a setup section built into the README.md.
But on a fresh RPi we need Tclx, as you write in eq3.exp (Note: requires package Tclx).
So the apt-get part needs to be:
`$ sudo apt install expect tclx8.4`

Greetings

Heckie75, 2017-11-05:
Dear SelmaUrban,

thanks for your feedback. Since I believe that the dependency to Tclx is not required anymore I have removed it from the script. 

It seems that everything works as expected. Nevertheless, if you think that the removal causes problems on Raspi, please let me know.

Best regards

SelmaUrban, 2017-11-05:
Hi,
with removed tclx8.4 and the new eq3.exp (without "package require Tclx") it works like before.

Heckie75, 2017-11-05:
Perfect! Thanks for testing on your Pi.

Best regards

closed 2017-11-06 by Heckie75