Heckie75/eQ-3-radiator-thermostat #6: JSON support

mbiernacik, 2017-12-19:
Hi @Heckie75,

It would be great to have support for JSON response from thermostat instead of plain text.
Thanks to that it would be much easier to parse data in external applications (eg. to visualize some data).

I don't know tcl/expect stuff, so it's hard for me to make a pull request with this change.
What do you think about it?

Cheers,
Marcin

Heckie75, 2017-12-20:
Dear Marcin,

please check my latest commit.

It works as follows:
```
$ ./eq3.exp 00:1A:22:0A:91:CF json
{
  "mac" : "00:1A:22:0A:91:CF",
  "temperature" : 20.0,
  "valve" : 0,
  "mode" : {
    "auto" : true,
    "manual" : false,
    "low battery" : false,
    "open window" : false,
    "vacation" : true,
    "locked" : false,
    "boost" : false,
    "unknown" : false,
    "dst" : false,
    "on" : false,
    "off" : false
  },
  "vacation" : "2017-12-20 15:00",
  "timers" : {
    "Sat" : [
      {
        "from" : "00:00",
        "to" : "07:30",
        "temperature" : 16.5
      },
      {
        "from" : "07:30",
        "to" : "11:00",
        "temperature" : 20.5
      },
      {
        "from" : "11:00",
        "to" : "15:00",
        "temperature" : 18.5
      },
      {
        "from" : "15:00",
        "to" : "21:00",
        "temperature" : 20.5
      },
      {
        "from" : "21:00",
        "to" : "24:00",
        "temperature" : 16.5
      }
...
```

Best regards,
Martin

mbiernacik, 2017-12-20:
Hi Martin,

it looks awesome! That's what I was thinking about it.
Thanks a lot ;)

I'll try to use this data in Domoticz or try to set timers based on Google Calendar.

Cheers,
Marcin

closed 2017-12-20 by mbiernacik