Heckie75/eQ-3-radiator-thermostat #8: Actual Temperature Reading

Misiek1977, 2018-01-18:
Hello 
One more question is it possible to get the actual temperature from the thermostat not just target one?
Anyway
Great Job

Michal

Heckie75, 2018-01-19:
Dear Michal,

unfortunately the thermostat itself does not have this possibiliy. Therefore it is not possible to get this value. 

Best regards,
Martin

Misiek1977, 2018-01-19:
It's unfortunate... So in order to connect it to my system I will have to add external temperature sensor.

Thanks for response...
Michal

thomasnordquist, 2018-01-20:
To be honest, measuring the temperature this close to the heater won't tell
you much about the room temperature. Get a proper temperature sensor and
place it somewhere in the room.