Heckie75/eQ-3-radiator-thermostat #10: Feature: Get Serial

sunflower-seed, 2018-02-12:
Hey,

thanks for your work.

i'm also digging in eq3 stuff.
I decompiled the App and found the PIN calculation is made with some XORs from device serial (well...nothing to say).
The serial is the String you can find under at the bottom of your device between the two batteries.

So I guess there is somehow a command to derive the serial from the device as an ASCII byte Array.
If i'm right the command byte is just one byte = 0, which return the device Information.

Maybe you could validate that and implement the Serial Output  into your script.

Heckie75, 2018-02-19:
Hi, 
actually I don't know for what reason the PIN is required. Therefore there is no use case. What I have seen so far is that the bluetooth pairing does not require any PIN - maybe I'm wrong here. 

If you give me an example calculation for PIN and an example how to request the serial number serial number via bluetooth I will add some functionality if it can be used for something.

Best regards,
Martin

jhmgbl, 2018-02-19:
> And how do I calculate the PIN?

I cant connect to the thermostat after successfull pairing. I always get a
timeout error!

sunflower-seed, 2018-02-20:
I tried to set up a secured BLE connection but this failed.
i guess the PIN should only prevent Users to connect to the wrong device (e.g. Neighbours), otherwhise they wont fetch the serial in the App and then do this PIN calculation Stuff.
Maybe they thought that everyone uses the App ;-)

I uploaded a quick serial calculation, that's more or less a copy of the original Android Code:
https://github.com/sunflower-seed/cc-rt-ble-pincheck

Feel free to implement this stuff, your tool is really helpful for Debugging my project using an esp32.
And I think a least knowing the pin makes User Experience in your own Application a bit nicer and you can detect the device you are talking to.

closed 2018-02-23 by Heckie75