Heckie75/eQ-3-radiator-thermostat #11: Feature: get current temperature

Equidamoid, 2018-03-03:
Is it possible to get the current temperature measured by the thermostat's sensor?
It will be pretty useful to calibrate the "offset" value from time to time.

I doubt the value is really exposed via BT since it is not visible in the official app, but maybe someone has an idea how to do it.

thomasnordquist, 2018-03-03:
They don't show the readout of the temperature in the app, because it would
not necessarily represent the room temperature. (Due to the proximity to
the heater)

The offset is a tool to calibrate the eq3 for the room size. The idea is
that you set the thermostat to a specific temperature, then you measure the
temperature at any point in your room (your couch or bed for instance). The
difference of your target temperature and the measured temperature is your
offset. The readout would not help you in this instance.

For most conceivable use cases the temperature sensor is too close to the
heater and would't be of any real value.

jhmgbl, 2018-03-03:
how do i pair the Thermostat i paired it after the instruction but i get no
connection

Equidamoid, 2018-03-03:
@thomasnordquist 
> The readout would not help you in this instance.
Why? I want to take the difference between the reading from the thermostat's sensor and a sensor in the middle of the room, average it over time a bit and set it as an offset. 
Basically doing exactly what you (and the manual for the thermostat) say, but automatically. 

// Ideally I'd like to directly set the valve position, but that's a different story.

Heckie75, 2018-03-04:
The bluetooth API does not seem to support reading room's temperature. Please take a look at the [eQ-3 radiator thermostat API](eq-3-radiator-thermostat-api.md)

That's probably why it isn't displayed in the app neither.

Equidamoid, 2018-03-04:
@Heckie75 Yup. I read it before posting here. And now going through that huge thread on FHEM forum I even found some cited responses from the manufacturer that they can't (more like don't want to) provide the data via BT.

closed 2018-04-02 by Heckie75