Heckie75/eQ-3-radiator-thermostat #14: Feature: manual control of valve opening percentage

garret, 2018-10-07:
First of all I wanted to thank you a lot for this script, you really did an amazing job!  

Second, I wanted to ask if there is a way to manually control how much the valve is opened and closed. I just bought these radiator valves and before I had the classical manual ones where you could regulate manually from 1 to 5 how much the valve was open. Is there a way to achieve something similar instead of reasoning in having the room to reach a certain temperature?

Heckie75, 2018-10-08:
Dear Garret, 

Since the script is limited to the capabilities of the thermostat and its provided API, this is not possible. The API has just the ability to set the temperature. 

Best regards

garret, 2018-10-10:
Thank you for your answer. So essentially there is a temperature sensor in the valve and, after I set a specific temperature, the valve starts the heating until that temperature is reached? 

May I ask what is the difference between the "auto" and "manual" option?

Heckie75, 2018-10-11:
"auto" runs the programs that you have set by using timers, e.g. set 20°C at 8:00 a.m., set 18.5°C at 10 a.m.

After you have switched to "manual" the thermostat just keeps the current temp-level and doesn't switch temp according programmed times and temperature. 

Best regards

garret, 2018-10-12:
Ok that is very clear! But at the same time, where I can see what is the current temperature measured by this radiator valve? 

I am curious to see how much that is influenced by the heat. It seems I can only set a temperature but no current temperature value is accessible.

Heckie75, 2018-10-12:
That's true. You can just see the target temperature that has been set. But there is no way to get the measured temperature. That's also a limitation of the device.

closed 2018-10-23 by Heckie75