Heckie75/eQ-3-radiator-thermostat #15: Connection failed

woemike, 2018-10-11:
Hello! I am using a Raspberry PI 3 Model B+ and I cannot connect to the thermostat. I run ./eq3.exp 54:4A:16:5C:9E:2C and it says connection failed. The error is: GDBus.Error.bluez.Error.AuthenticationFailed: Authentication Failed. Thank you!

Heckie75, 2018-10-12:
Maybe the device must be paired with the raspi before you can use it. See chapter 2 in README file.

Pairing can be done on console like this:

1. use "bluetoothctl"
2. turn "agent on"
3. turn "scan on"
4. wait until the mac address of the thermostat appears
5. enter "pair 54:4A:16:5C:9E:2C"

I hope that this helps. Are there any other users kn

Best regards

woemike, 2018-10-12:
Hello! Those are the exact steps I am following and nothing.

Heckie75, 2018-10-13:
Do you try to pair the device with mac address  54:4A:16:5C:9E:2C ?

As far as I can see in your screenshot this is a "comet blue" device. My script is not made for this. You need to have a thermostat by eq-3 / Eqiva which is the CC-RT-BLE-EQ.

Best regards

closed 2018-10-23 by Heckie75

vcmorini, 2019-05-24:
Dear Heckie75, is pairing necessary to read/write? 
Thks!

vcmorini, 2019-05-24:
Dear Heckie75, is pairing necessary to read/write? 
If yes, this would almost make any iot prj unfeasible, since pressing the button of eq-3 is uncomfortable. 
Do you have any suggestion or workaround? 
Thks a lot

janvda, 2019-05-24:
Hi, I don't think you need to pair each time.

Anyhow I have use the following https://github.com/janvda/balena-node-red-eq3 to connect a raspberry pi to a single eq-3 device and I don't need to "pair" it manually each time to read/write.
This repository is based on node-red-eq3 which is based on the reversed engineered protocol by Heckie75.