Heckie75/eQ-3-radiator-thermostat #16: Cannot read timers

doidor, 2018-10-22:
There seems to be an issue with reading timers on these thermostats. The timers command throws the following error:

`./eq3.exp 00:1A:22:06:31:84 timers
can't read "d": no such variable
    while executing
"return $d"
    (procedure "hexToDecimal" line 3)
    invoked from within
"hexToDecimal [lindex $raw_timer [expr 2 * $event + 1 ]]"
    (procedure "readTimer" line 13)
    invoked from within
"readTimer $day"
    (procedure "readTimers" line 5)
    invoked from within
"readTimers"
    (procedure "init" line 14)
    invoked from within
"init $INIT(timer)"
    ("^timers$" arm line 2)
    invoked from within
"switch -regexp $c {
    ^sync$ {
      init $INIT(mode)
    }
    ^comforteco.* {
      checkArgs $c 2 "comforteco" {^comforteco [0-9]{1,2}\.?[05]? [0..."
    (procedure "doCommand" line 10)
    invoked from within
"doCommand"
    (file "./eq3.exp" line 1112)`

I cannot access the weekly schedule from the calor BT app either. Has anyone else encountered this?

Cheers

Heckie75, 2018-10-23:
Hi Doidor, 

this error is raised by the script after the thermostat hasn't send a response in time. SInce the script sends the same commands to the thermostat that also the app sends, the behaviour is simuluar. Actually I think that your thermostat is broken. I had several of these devices and returned at least two of them for simular reasons either right from the beginning or after a couple of days. 

For further analysis you can do the following. You can set the script into debug mode by editing it. Change the following line:

log_user 0

to 

log_user 1

You will get more information now. You will probably notice that the thermostat is not responding after try to get timer information. 

In addition you can try to increase the timeout value. Therefore you can change the following line:

set timeout 10

to 

set timeout 20

where 20 means 20 seconds. 

Best regards,
Martin

doidor, 2018-10-23:
Hey,

I've actually factory reset (using the reset param you provided) all my thermostats, repaired them and everything went well after that.

Cheers,
Tudor

Heckie75, 2018-10-23:
Perfect!

closed 2018-10-23 by Heckie75

eXicRa, 2018-11-08:
Thank you for your great script.
I had the same error, the resetting of the thermostat helped me also.