Heckie75/eQ-3-radiator-thermostat #18: Timeout when reading status

ccrisan, 2018-12-02:
I can't seem to read status from my eQ3. It always says timeout:

    ccrisan@asus430ua:~/downloads/eQ-3-radiator-thermostat-master$ ./eq3.exp 00:1A:22:05:D1:17 status
    ERROR: Thermostat hasn't responded after sync request in time (10 sec.)

The weird thing is that other functions, such as *lock*, seem to work, even though I keep getting the timeout; I see the `LOC` text appearing on my thermostat when I run:

    ccrisan@asus430ua:~/downloads/eQ-3-radiator-thermostat-master$ ./eq3.exp 00:1A:22:05:D1:17 lock
    ERROR: Thermostat hasn't responded in time (10 sec.)

I tried digging deeper into this problem and used `gatttool` manually (I need to interrupt it after waiting more than 1 minute):

    ccrisan@asus430ua:~$ gatttool -b 00:1A:22:05:D1:17 --listen --char-write-req --handle=0x0411 --value=03181202151f05
    Characteristic value was written successfully
    ^C

A few remarks:

 * this happens for 3 out of 8 eQ3 thermostats I have installed in my house
 * I tried updating the firmware via the official app to `1.20` - same result
 * I (factory) reset the device many times without any success

Any help would be appreciated.

Heckie75, 2018-12-11:
Hi,

Ich have read Simulation Situationen before. Last time somebody wrote that he performed a factory reset by the script like ./eq3.exp 00:1A:22:05:D1:17 reset

In Addition I recommend to use gatttool in interactive mode when Debugging.

ccrisan, 2018-12-12:
@Heckie75 I have tried the `reset` command and it doesn't work. I have manually reset the thermostat from its menu a couple of times with no success.

I am also using `gatttool` in interactive mode when debugging but the results are the same - no notifications.

ccrisan, 2018-12-13:
@Heckie75 I have managed to reset the thermostat using `./eq3.exp <MAC> reset` - the same effect as when reset from the menu.

The seems seems to work (maybe because it doesn't need the responses from the thermostat?).

I have also tried pairing using `bluetoothctl` and it worked perfectly.

Heckie75, 2018-12-13:
Since you have written that you 'tried using bluetoothctl and it worked perfectly' that the script run after you have paired the device?

Actually the app also needs the responses. Otherwise it wouldn't be alble to display status like temperature, auto or manual mode etc.

ccrisan, 2018-12-13:
The pairing worked perfectly but the script still does not receive notifications and neither does a manual run of `gatttool`.

Heckie75, 2018-12-18:
Maybe another try since I just ran into a simular situation where the thermostat didn't answer. 

Press and hold the button in front of the thermostat until "Pair" is displayed. Thermostat is in pairing mode. 

Then  perform ``` ./eq3.exp <MAC> serial```.

In my case I was able again to perform a ``` ./eq3.exp <MAC> serial``` and the thermostat answered again.

ccrisan, 2018-12-18:
@Heckie75 thanks for getting back to me. I have tried your commands but unfortunately the problem stays the same:

    ./eq3.exp 00:1A:22:05:D1:17 serial

    Serial no.:		MEQ0409228
    Pair:			2685

And then:

    ./eq3.exp 00:1A:22:05:D1:17 sync
    ERROR: Thermostat hasn't responded after sync request in time (10 sec.)

And then again:

    ./eq3.exp 00:1A:22:05:D1:17 status
    ERROR: Thermostat hasn't responded after sync request in time (10 sec.)

ccrisan, 2018-12-19:
After starting to have the problem, I have upgraded one of my thermostats to 1.2. I now have 3 thermostats exhibiting the problem: 2 with 1.1 and 1 with 1.2.

I will try the BT sniffing method and see what the communication looks like when working with the app.

luckyhacky, 2018-12-30:
Hi all, thanks for the great project.
I had the same issue and did some research on my own.

> Taking a virgin device:

Connecting to BTLE device is working with gattool.
Connecting with ./eq3.exp results in timeout
Pairing on linux and connecting with ./eq3.exp results in timout

> Install device on radiator and do init sequence, remove device, pairing with smartphone

Connecting with ./eq3.exp is possible if not connected under linux (connection will be established by ./eq3.exp)
Connecting with ./eq3.exp is not possible if connected under linux, e.g. with bluez

> More info

Linux does not require a pin on my laptop ???
"calor BT" app requires a pairing pin

> Result
it seems, that the EQ3 needs to have run a complete setup, then BTLE connection is possible.
Can other confirm this?

ccrisan, 2019-06-11:
The latest thermostat firmware (version reported is 1.2) fixed this issue for me on all my thermostats that were affected.

closed 2019-06-11 by ccrisan