Heckie75/eQ-3-radiator-thermostat #20: Antwortet nicht

Oemer9119, 2018-12-12:
Hallo,
erstmal auch vielen dank für das programm.
ich habe es oft versucht aber kriege leider keine antwort.
ich kann mich verbinden und die befehle senden die nimmt er auch an und ändert die temperatur auf dem display. aber weder bei status noch bei char-write-req 0411 00 bekomme ich eine Antwort.
fehler meldung kommt aber auch nicht. 
danke im vorraus.

Heckie75, 2018-12-13:
Hallo,

klingt ähnlich wie das was ich gerade in https://github.com/Heckie75/eQ-3-radiator-thermostat/issues/18 beantwortet habe.

Funktioniert das Thermostat denn mit der Android App? Ich habe einen Thermostat zuhause, der von anfang an nicht via Bluetooth funktionierte. Gleiches Verhalten und funktioniert dann natürlich auch nicht in der App. 

Wenn die App funktioniert, muss es ja an der Verbindung zum PC liegen. Dann könnte ich mir noch vorstellen, dass u.U. doch ein Pairing erforderlich ist, also: 

* "bluetoothctl" aufrufen
* "agent on" einstellen
* "scan on" aktivieren
* pair <mac> durchführen

Neulich gab es eine Aktualisierung der Firmware. Dies hatte aber keinen Einfluss auf die Funktionsweise.

Oemer9119, 2018-12-21:
Hallo, 
ich habe die Schritte befolgt.
Mit der App klappt alles, ist war nicht grad das beste aber ist ya ein anderes Thema.

[image]
[image]

nach den Befehlen reagiert das Thermostat auch ohne probleme, aber halt ohne Antwort.

Heckie75, 2018-12-21:
Hallo,

das Problem ist offenbar ident. mit https://github.com/Heckie75/eQ-3-radiator-thermostat/issues/18 

Da ich es selbst nicht reproduzieren kann, hoffe ich auf das Feedback in dem genannten Issue. Vielleicht finden wir dort eine Lösung. 

Gruß

Oemer9119, 2018-12-21:
Hallo,

leider klappt das auch nicht. Werde mir mal noch eins Bestellen und dort das gleiche versuchen und dann vergleichen.
[image]

Oemer9119, 2019-01-01:
Hallo und frohes neues Jahr.
Ich habe Festgestellt, dass eigentlich alles Funktioinert bis ich ioBroker Starte.
In Putty bekomme ich eine Antwort aber wenn ioBroker noch nicht gestartet ist oder nach dem es Beendet wurde.

Ich teste mal wenn ich den Script und alles Lösche, ob es dann tzrozdem eine Auswirkung hat.

ccrisan, 2019-01-04:
@Darkexy as far as I know, the thermostat can only be connected to one single bluetooth adapter at the time. The behavior that you see is normal.

Darkexy, 2019-01-04:
> 
> 
> @Darkexy as far as I know, the thermostat can only be connected to one single bluetooth adapter at the time. The behavior that you see is normal.

Thanks for your answer. Well, I'm not totally into the whole BLE thing, but as long as the BLE-Adapter in ioBroker doesn't show the thermostat, it isn't actually connected, or am I wrong? I don't understand, why the gattool and the script won't work while the BLE-Adapter is running, but works, if i turned it off, if it is not really connected?

mweisshaupt1988, 2019-12-19:
Ich habe ein ähnliches Problem, bei mir scheint aber die Reichweite die Ursache zu sein. In 2 Meter Entfernung funktioniert mit dem Skript alles aber in 4 Metern nicht mehr. Ich habe den Bluetooth Stick "CSL - Bluetooth 4.0 USB Adapter" und vermute, dass es an dem liegt weil ich mit dem Handy auch am anderen Ende der Wohnung eine Verbindung zu dem Thermostat aufbauen kann.

Gibt es Erfahrungen mit welchen Sticks die Verbindung gut funktioniert?

closed 2020-02-16 by Heckie75