Heckie75/eQ-3-radiator-thermostat #21: $1 Variable setzen für Skriptautomation

JsBergbau, 2019-01-22:
Der Equiva soll als Frostschutz dienen. Frostschutz wird ausgelöst über ein Skript in Abhängigkeit von der Außentemperatur und der Temperatur am Vorlauf des Heizkörpers via Sensor. Das Skript kann nach Fertigstellung gerne zur Verfügung gestellt werden. Habe das Thermostat erstmal in der App gepaart. Am Raspberry war danach keinerlei Pairing erforderlich, es kann einfachso gesteuert werden. Vielleicht verliert sich dieses Pairing einfach so. Da es ein Skript steuert und damit der Frostschutz funktioniert, wollte ich via $1 überprüfen, ob Fehler aufgetreten sind. Leider ist $1 immer leer, egal ob der Befehl erfolgreich war oder fehlschlug. Daher mein Featurerequest einfach $1 entsprechend setzen, 0 bei keinem Fehler, irgendwas anderes bei Fehler.

Zocker1999NET, 2019-11-16:
Meinst du eigentlich nicht den Exit Code, einzusehen nach POSIX unter `$?`? `$1` steht (zumindest unter POSIX) für den ersten Parameter beim Aufrufen eines Shell-Skripts

Heckie75, 2020-02-16:
Korrekt, dies wäre am Exit-Code festzumachen.

Beispiel:
```
$ eq3.exp 00:1A:22:0A:91:CF status

Temperature:                    21.0°C
Valve:                          0%
Mode:                           auto dst
Vacation mode:                  off

...

$ echo $?
0
```

closed 2020-02-16 by Heckie75