Heckie75/eQ-3-radiator-thermostat #23: Implementierung von Retries

Tomatenjoghurt, 2019-11-28:
Hallo zusammen,

ich hab das Script im Einsatz. Leider ist ein Thermostat im Haus relativ weit entfernt und so hat der Pi nur sporadisch eine Verbindung zum Gerät. Die Erfahrung zeigt allerdings, dass nach 2-3 Versuchen eine Verbindung zu stande kommt. Gibt es eine Möglichkeit eine Art Globalen Parameter "maxRetrys" zu implementieren?
Kenne mich leider mit Shell-Scripting 0 aus...

Heckie75, 2020-02-16:
Dieses Script könnte das Problem lösen. Lege es am besten in das gleiche verzeichnis wie das ```eq3.exp```-Skript 

```
#!/bin/bash
MAX_RETRIES=3

retry=0
while [ $retry -lt $MAX_RETRIES ]
do
  eq3.exp "$@"
  if [ $? == 0 ]
  then
    exit 0
  fi
  retry=$(($retry+1))
  sleep 1
done
echo "FAIL"
exit 1
``` 
Kopiere das Script in eine Datei, z.B. "eq3_retry" und gebe der Datei Ausführungsrechte:

```
chmod +x eq3_retry
```

PS: Falls das Skript meckert, dass ```eq3.exp``` nicht gefunden werden kann, ist das ```eq3.exp``` wahrscheinlich nicht im Pfad.

closed 2020-02-16 by Heckie75

Tomatenjoghurt, 2020-02-17:
So simpel und doch genial - hätte ich auch selbst drauf kommen können. Vielen Dank, werde das die Tage einmal einbauen!