Heckie75/eQ-3-radiator-thermostat #25: Security concerns - Connection without PIN/Pairing

agapeDi, 2020-01-07:
Hello, this is rather a question than a real issue. Hope someone could still share thoughts. 

I can use the device from my Raspbian and was able to control the temperature. The device didn't need to be paired, nor did I have to enter the PIN somewhere - like it's mentioned also in the documentation and in other issues. Now I'm a bit concerned about my security, I fear people could drive-by my house and control my radiators from outside. Since I'm not a Bluetooth expert I was wondering whether that is even possible through a wall with the EQ3 device? 

Also I updated to latest firmware 1.2 from my mobile phone and was hoping that with that it requires at least a pairing, but still I can control the device without pairing and without PIN...

Does anyone know a more secure alternative device which can also be easily controlled with a Raspberry Pi from the command line? It seems a lot of the devices need additional radio adapters (such as Homematic, Zigbee, etc) or they have no documented interface or work only via some third-party cloud services. 

Thanks a lot for your help!

Heckie75, 2020-02-16:
Hi, 

your concerns in terms of security are fair. I have reengineered many bluetooth LE devices and have seen that no device has serious security features. 

In most cases the supposed security features like PIN are implemented on side of the app but not on side of the device.  For example, in case of the EQ-3 device the app asks for the serial number of the radiator thermostat. Based on the serial number the app calculates the PIN and compares it to the PIN that you have entered. If the entered PIN was correct the app continous operation. Actually this process would have been done on device's side, so that the device blocks requests if PIN was not correct. But that's obviously not the case. In case of other devices the PIN can be reset to a default value without sending the old PIN.

Bluetooth LE doesn't seem to have security features. For example, there is no need of pairing devices. Therefore security has to be implemented on application level instead of protocol level. I guess that microcontrollers, that are used in these devices such as the radiator thermostat, don't have enough capacity to calculate complex algorithms that guarantee reasonable security. That's probably the reason for implementing it on client's site - in the app.

To be be honest, I'm not very afraid of being hacked. Of course, somebody could heat my flat to 30°C or turn my lights on (see Mipow Playbulbs). But the probability seems to be very low. 

Cheers

closed 2020-02-16 by Heckie75

JsBergbau, 2020-11-29:
@agapeDi:
I have the same concerns as you. So one solution is to query your thermostats in regulary intervals like every hour or so. Then check the setting if it is in a reasonable state like temperature between 17 and 22 degrees. If not correct it and raise an notification like E-Mail, that there is a possibly attack.