Heckie75/eQ-3-radiator-thermostat #26: eQ3 - Connection failed very often

JacekKac, 2020-10-03:
raspberry pi 4b  eQ-3 CC-RT-BLE. 

2 eQ-3 devices, paired and working, but frequently i get Connection failed (25% of tries)

1/ Connenction Failed, btmon log during connection:

HCI Event: Command Complete (0x0e) plen 14                                                                                                     #425 [hci0] 172.457975
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
        00 00 00 00 00 00 00 00 00 00                    ..........
< HCI Command: LE Add Device To White List (0x08|0x0011) plen 7                                                                                  #426 [hci0] 172.458043
        Address type: Public (0x00)
        Address: 00:00:00:00:00:00 (OUI 00-00-00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                      #427 [hci0] 172.458441
      LE Add Device To White List (0x08|0x0011) ncmd 1
        Status: Repeated Attempts (0x17)
> HCI Event: LE Meta Event (0x3e) plen 12                                                                                                        #428 [hci0] 172.766608
      LE Read Remote Used Features (0x04)
        Status: Connection Failed to be Established (0x3e)
        Handle: 64
        Features: 0x3f 0x00 0x00 0x08 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Slave-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
          Unknown features (0x0000000008000000)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                   #429 [hci0] 172.767179
        Status: Success (0x00)
        Handle: 64
        Reason: Connection Failed to be Established (0x3e)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                            {0x0002} [hci0] 172.804587
        LE Address: 00:1A:22:11:80:74 (eQ-3 Entwicklung GmbH)
        Reason: Unspecified (0x00)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                            {0x0001} [hci0] 172.804587
        LE Address: 00:1A:22:11:80:74 (eQ-3 Entwicklung GmbH)
        Reason: Unspecified (0x00)

2/ Second problem - script sometimes is giving an unexpected result:

pi@raspberrypi:~ $ /opt/eQ-3-radiator-thermostat/eq3.exp 00:1A:22:11:80:74 json
^[[6~can't read "d": no such variable
    while executing
"return $d"
    (procedure "hexToDecimal" line 3)
    invoked from within
"hexToDecimal [lindex $raw_timer [expr 2 * $event + 1 ]]"
    (procedure "readTimer" line 13)
    invoked from within
"readTimer $day"
    (procedure "readTimers" line 5)
    invoked from within
"readTimers"
    (procedure "toJson" line 14)
    invoked from within
"toJson"
    ("^json$" arm line 2)
    invoked from within
"switch -regexp $c {
    ^sync$ {
      init $INIT(mode)
    }
    ^comforteco.* {
      checkArgs $c 2 "comforteco" {^comforteco [0-9]{1,2}\.?[05]? [0..."
    (procedure "doCommand" line 10)
    invoked from within
"doCommand"
    (file "/opt/eQ-3-radiator-thermostat/eq3.exp" line 1157)

is it normal? maybe someone is experiencing the same issue ?

pako-github, 2020-10-12:
I am experiencing similar problem. I even cannot connect to one of my thermostats anymore although it is no problem to connect it with the phone app. Tried to increase the timeout time which didn't work. But if I do that the debug log says that the connection was refused .

janvda, 2020-10-12:
I remember that raspberry pi Bluetooth range is not huge.  So it might be a bluetooth range issue.
Are you getting the same issue when putting the raspberry really close to the eQ3 device ?

pako-github, 2020-10-12:
Yes, I can exclude range issues. It worked pretty for a certain amount of time and then it stopped. Hence it could also be a problem with my Linux. Debugging Bluetooth is not always easy...

whissperr, 2020-11-19:
Same connection issues with Raspberry pi 4b and eQ-3 CC-RT-BLE starting from September. Before had no problems. All devices are permanently fixed, so can exclude bluetooth range issue.

JsBergbau, 2020-11-29:
Please have a look at https://github.com/abandonware/noble/issues/99 and https://github.com/RPi-Distro/bluez-firmware/issues/6
For me doing it manually with gatttool from the RPi4 it worked all the time (about 5 times). 
But still the RPi4 has really a lot of problems with bluetooth, see also https://github.com/RPi-Distro/firmware-nonfree/issues/8

whissperr, 2020-11-30:
> Please have a look at [abandonware/noble#99](https://github.com/abandonware/noble/issues/99) and [RPi-Distro/bluez-firmware#6](https://github.com/RPi-Distro/bluez-firmware/issues/6)
> For me doing it manually with gatttool from the RPi4 it worked all the time (about 5 times).
> But still the RPi4 has really a lot of problems with bluetooth, see also [RPi-Distro/firmware-nonfree#8](https://github.com/RPi-Distro/firmware-nonfree/issues/8)

Thank you very much! Looks like updating firmware as suggested worked in my case. Connection is more stable than before.

JacekKac, 2020-12-06:
> Thank you very much! Looks like updating firmware as suggested worked in my case. Connection is more stable than before.

Hi, can you point exact steps how to update the firmware? there are 3 different links and a lot of posts, which way worked for you ?

whissperr, 2020-12-08:
> Hi, can you point exact steps how to update the firmware? there are 3 different links and a lot of posts, which way worked for you ?

Sorry, unfortunately after some days doing tests, can not confirm that solution is really working for me. You can try on your system, I've replaced original firmware with the one from this post: https://github.com/RPi-Distro/bluez-firmware/issues/6#issuecomment-732027094.

JsBergbau, 2020-12-08:
I can recommend this bluetooth dongle https://www.aliexpress.com/item/32696756077.html
Quite cheap but working great for me. At least range is similar or higher than built in from Raspberry PI 3B.

To use this interface, you have to modify the script. Details see here https://github.com/Heckie75/eQ-3-radiator-thermostat/issues/30

JacekKac, 2020-12-12:
@whissperr  worked form me. BT problems are gone

https://www.raspberrypi.org/forums/viewtopic.php?t=292113&p=1766154

JacekKac, 2020-12-12:
BT firmware upgrade solved BT connection issues and wifi. Everythink in me pi works flawless right now, it was never a eQ-3 problem

closed 2020-12-12 by JacekKac