Heckie75/eQ-3-radiator-thermostat #29: Additional values of handle 0x0421 in new versions, what do they mean?

JsBergbau, 2020-11-29:
Hallo,

a few years old device
```
char-write-req 0411 4500
Characteristic value was written successfully
Notification handle = 0x0421 value: 02 01 09 50 04 0c
```

New device ordered a few weeks ago:
```
char-write-req 0411 4500
Characteristic value was written successfully
Notification handle = 0x0421 value: 02 01 09 50 04 1e 00 00 00 00 18 03 2a 22 07
```

So anyone has an idea what these new values `00 00 00 00 18 03 2a 22 07` mean?

llemtt, 2020-11-30:
My eq3's are from 2018 and iirc they always sent those "extra" bytes, still don't know what they means.

Regarding "valve" current temperature I remember a statement from Eqiva on their IP products saying that they don't report it because it's meaningless...  I just bought a bunch of inexpensive Xiaomi thermometers.

Heckie75, 2020-12-26:
Hi,

after some investigation I found the following:

```
Notification handle = 0x0421 value: 02 01 09 50 04 1e 00 00 00 00 18 03 2a 22 07
                                    |  |  |  |  |  |  |  |  |  |  |  |  |  |  + (unknown)
                                    |  |  |  |  |  |  |  |  |  |  |  |  |  + eco temperature (devid value by 2)
                                    |  |  |  |  |  |  |  |  |  |  |  |  + comfort temperature (devid value by 2)
                                    |  |  |  |  |  |  |  |  |  |  |  + open windows interval in 5 minute steps 
                                    |  |  |  |  |  |  |  |  |  |  + temperature in open windows mode (devide value by 2) 
                                    |  |  |  |  |  |  |  |  |  + if vacation mode: time in 30 minutes steps
                                    |  |  |  |  |  |  |  |  + if vacation mode: month (January = 0, February = 1, etc.)
                                    |  |  |  |  |  |  |  + if vacation mode: year (add 2000)
                                    |  |  |  |  |  |  + if vacation mode: day in month
                                    |  |  |  |  |  + target temperature (devide value by 2)
                                    |  |  |  |  + (unknown)
                                    |  |  |  + Valve in percent 
                                    |  |  + Mode as already described in API
                                    |  + 0x01 if this notification is device status notification
                                    + Always 0x02 if this notification is device status notification
```

Maybe I will add these values to the script later. 

Cheers

closed 2020-12-26 by Heckie75

llemtt, 2020-12-28:
> Hi,
>
> after some investigation I found the following:
>
> ```
> Notification handle = 0x0421 value: 02 01 09 50 04 1e 00 00 00 00 18 03 2a 22 07
>                                     |  |  |  |  |  |  |  | [...]

07 (the last byte) is the "temperature offset"

cheers

Heckie75, 2020-12-28:
Thanks, I have noticed this and enabled the script to print these additional information like this:

```
$ eq3 K sync

Temperature:			16.5°C
Valve:				0%
Mode:				auto dst
Vacation mode:			off

Comfort temperature:		20.5°C
Eco temperature:		17.5°C

Open window temperature:	4.5°C
Open window time:		00:30

Offset temperature:		0.0°C
```