Heckie75/eQ-3-radiator-thermostat #31: No paring required

JsBergbau, 2020-12-14:
> Actually I am not 100% sure, but it seems that the thermostat must be explicitly paired (maybe only newer versions). Therefore you have to press the button on the termostat and start the pairing procedure. It depends on your linux distribution how devices must be paired there. The pin seems not to be required for pairing.

Paring is not required. However after inserting battery you have to disable and re-enable bluetooth in menu. Otherwise device is not found via bluetooth. After that you can immediately control it via BT.

TazerReloaded, 2020-12-22:
Can confirm, got mine today, no pairing required, just scanned for the MAC and was up and running. But maybe it starts pairing mode automatically after power on for some time, I'll wait a few days and try again. Seems a bad design to pair with anything, I don't want my neighbor to set my room temperature.

0xFACE, 2020-12-23:
Recently there was a firmware upgrade released and now pairing is required (6 digit pin).

kjyv, 2020-12-23:
I paired my Raspberry with it but can't seem to connect ("Connection failed.") afterwards. Without pairing, I got this 
```
expect: spawn id exp4 not open
    while executing
"expect -re "\[0-9a-f\ ]*" {
    log "ok"
  }"
    (procedure "readHandle" line 10)
    invoked from within
"readHandle $HANDLES(vendor)"
    (procedure "readDeviceInfo" line 6)
    invoked from within
"readDeviceInfo"
    (procedure "init" line 8)
    invoked from within
"init [expr $INIT(device) + $INIT(mode) + $INIT(timer) + $INIT(serial)]"
    ("^status$" arm line 2)
    invoked from within
"switch -regexp $c {
    ^sync$ {
      init $INIT(mode)
    }
    ^comforteco.* {
      checkArgs $c 2 "comforteco" {^comforteco [0-9]{1,2}\.?[05]? [0..."
    (procedure "doCommand" line 10)
    invoked from within
"doCommand"
    (file "./eq3.exp" line 1156)
```

Heckie75, 2020-12-26:
I think that your problem is, that the `expect` session was closed unexpectedly.

Please try if you can run the following on command line:
```
$ gatttool -I -b 00:1A:22:0A:91:CF
[00:1A:22:0A:91:CF][LE]> connect
Attempting to connect to 00:1A:22:0A:91:CF
Connection successful
[00:1A:22:0A:91:CF][LE]> char-write-req 0411 4500
Characteristic value was written successfully
Notification handle = 0x0421 value: 02 01 08 00 04 1e 00 00 00 00 0a 02 2a 23 07
[00:1A:22:0A:91:CF][LE]>
```

Don't forget to replace the MAC address with yours.

kjyv, 2020-12-26:
Sorry, should have updated that it worked fine after updating the bluetooth firmware of my raspberry pi (and restarted) - not sure which of the two fixed it. My comment was intended to show that there was a difference in pairing and not pairing though.

Heckie75, 2020-12-27:
Ok, perfect!

closed 2020-12-27 by Heckie75