Heckie75/eQ-3-radiator-thermostat #32: new firmware version added security?

llemtt, 2020-12-23:
Hi, a new app version is available with a new firmware release with improved security features,

anyone has already tried it?

kjyv, 2020-12-23:
I haven't tried it with the old version but I can't connect and I'm running the new version, so could be related

kjyv, 2020-12-23:
Now, not really sure which one it is, but I've followed https://www.raspberrypi.org/forums/viewtopic.php?t=292113&p=1766154 to update the bluetooth firmware and then restarted the (which might or might not have included previous kernel updates) - works fine now with the new eQ firmware (1.46)

TazerReloaded, 2020-12-24:
I'm a bit hesitant to install any firmware updates, because if Bluetooth connection is lost while updating, the device is basically hard bricked forever (see Google Play Store user reports for examples). Furthermore, the new Android app (2020-12-21) does not work at all on my phone, it starts, but I'm not able to press any buttons. So I'll just leave mine as it is until I see some evidence that the new firmware is working properly, and hope that no one takes control from outside in the meantime.

JacekKac, 2020-12-24:
> I'm a bit hesitant to install any firmware updates, because if Bluetooth connection is lost while updating, the device is basically hard bricked forever (see Google Play Store user reports for [...]

Its very easy to unbrick this device if it would really happen so dont worry to update 👍 

https://diy.stackexchange.com/questions/104473/my-eq-3-bluetooth-radiator-thermostat-cc-rt-ble-is-just-showing-upd-how-can-i

Heckie75, 2020-12-26:
Please let me know if this script is still running after updating the firmware.

Actually I'm wondering how you can enter a 6 digit PIN since device can't display more than 4 digits.

0xFACE, 2020-12-26:
It displays the first 3 digits in sequence, followed by the next 3 digits.
Something like this:
123-
-456

Heckie75, 2020-12-26:
Does the script still work?

What is the output of the following commands?

```
$ gatttool -I -b 00:1A:22:0C:19:60
[00:1A:22:0C:19:60][LE]> connect
Attempting to connect to 00:1A:22:0C:19:60
Connection successful
[00:1A:22:0C:19:60][LE]> char-write-req 0411 4500
Characteristic value was written successfully
Notification handle = 0x0421 value: 02 01 0a 00 04 21 1c 14 10 0c
[00:1A:22:0C:19:60][LE]> characteristics
handle: 0x0110, char properties: 0x02, char value handle: 0x0111, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0120, char properties: 0x02, char value handle: 0x0121, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0130, char properties: 0x02, char value handle: 0x0131, uuid: 00002a02-0000-1000-8000-00805f9b34fb
handle: 0x0140, char properties: 0x08, char value handle: 0x0141, uuid: 00002a03-0000-1000-8000-00805f9b34fb
handle: 0x0150, char properties: 0x02, char value handle: 0x0151, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0210, char properties: 0x22, char value handle: 0x0211, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x0310, char properties: 0x02, char value handle: 0x0311, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x0320, char properties: 0x02, char value handle: 0x0321, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x0410, char properties: 0x0a, char value handle: 0x0411, uuid: 3fa4585a-ce4a-3bad-db4b-b8df8179ea09
handle: 0x0420, char properties: 0x1a, char value handle: 0x0421, uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a
handle: 0xff01, char properties: 0x38, char value handle: 0xff02, uuid: e3dd50bf-f7a7-4e99-838e-570a086c666b
handle: 0xff04, char properties: 0x08, char value handle: 0xff05, uuid: 92e86c7a-d961-4091-b74f-2409e72efe36
handle: 0xff06, char properties: 0x02, char value handle: 0xff07, uuid: 347f7608-2e2d-47eb-913b-75d4edc4de3b
```

0xFACE, 2020-12-26:
Yes, it works. But maybe my raspberry pi was paired before. So not 100% sure it will work for all cases ;-)
```
[00:1A:22:0C:xx:xx][LE]> connect
Attempting to connect to 00:1A:22:0C:xx:xx
Connection successful
[00:1A:22:0C:xx:xx][LE]>  char-write-req 0411 4500
Characteristic value was written successfully
Notification handle = 0x0421 value: 02 01 09 00 04 29 00 00 00 00 18 03 2a 22 07
[00:1A:22:0C:xx:xx][LE]> characteristics
handle: 0x0110, char properties: 0x02, char value handle: 0x0111, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0120, char properties: 0x02, char value handle: 0x0121, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0130, char properties: 0x02, char value handle: 0x0131, uuid: 00002a02-0000-1000-8000-00805f9b34fb
handle: 0x0140, char properties: 0x08, char value handle: 0x0141, uuid: 00002a03-0000-1000-8000-00805f9b34fb
handle: 0x0150, char properties: 0x02, char value handle: 0x0151, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0210, char properties: 0x22, char value handle: 0x0211, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x0310, char properties: 0x02, char value handle: 0x0311, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x0320, char properties: 0x02, char value handle: 0x0321, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x0410, char properties: 0x0a, char value handle: 0x0411, uuid: 3fa4585a-ce4a-3bad-db4b-b8df8179ea09
handle: 0x0420, char properties: 0x1a, char value handle: 0x0421, uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a
handle: 0xff01, char properties: 0x38, char value handle: 0xff02, uuid: e3dd50bf-f7a7-4e99-838e-570a086c666b
handle: 0xff04, char properties: 0x08, char value handle: 0xff05, uuid: 92e86c7a-d961-4091-b74f-2409e72efe36
handle: 0xff06, char properties: 0x02, char value handle: 0xff07, uuid: 347f7608-2e2d-47eb-913b-75d4edc4de3b
```

Heckie75, 2020-12-26:
So, since it is still working it would be interesting what the output of the following is and how it is related to the 6 digit PIN?

```
$ eq3.exp 00:1A:22:0A:91:CF serial

Serial no.:             OEQ0616414
Firmware:               1.2
Pair:                   4758
```

I hope that the 6 digit PIN is just the numeric part of the serial.

0xFACE, 2020-12-26:
```
 ./eq3.exp  00:1A:22:07:31:2D   serial

Serial no.:             MEQ0433298
Firmware:               1.46
Pair:                   2315
```

Requested pairing PIN for this device is 297313.

kjyv, 2020-12-26:
I paired mine after it was updated to the new firmware already and it also works alright.

Heckie75, 2020-12-27:
So, I have updated two devices to the new firmware 1.46 and I am still able to control these by using the script.

So it seems that paring or entering a PIN is still not required. I have no idea how the new PIN is generated. I guess that it is also based on the serial number or maybe the MAC address?

Here are some examples:

```
Device 1
- MAC:   00:1A:22:07:31:2D
- Serial: MEQ0433298
- Old PIN: 2315
- New PIN: 297313

Device 2
- MAC: 00:1A:22:0C:19:60
- Serial: OEQ1755226
- Old PIN: 3531
- New PIN: 913219

Device 3
- MAC: 00:1A:22:0A:91:CF
- Serial: OEQ0616414
- Old PIN: 4758
- New PIN: 849839
```

0xFACE, 2020-12-27:
I had issue with a PIN like 000-255. 
I had fail with firmware update with one of my device. 
I think root cause was weak battery during update - device rebooted before update was finished. 
After that device displayed PIN 000-255, but the official app asked me for 4 digit PIN. 

Anyway, thanks to your scrpit I found proper 4-digit pin to reconnect failed device and I was able to do a upgrade firmware again.
So for now it is fully recovered.

closed 2020-12-27 by Heckie75