Heckie75/eQ-3-radiator-thermostat #33: Lowercase in MAC-address as alias recognized, breaks compatibility

JsBergbau, 2020-12-28:
Hallo Heckie75,

thanks for updating your script and especially implementing using the script with other bluetooth interfaces than hci0
Implementing your alias feature I've noticed that the script thinks you have specified an alias when at least one letter is written in lowercase like
```
eq3.exp AA:BB:CC:DD:EE:Ff sync
device "AA:BB:CC:DD:EE:Ff" isn't known yet. Please check ~/.known_eqivas file before using alias.
```
In old version lowercase mac-address was no problem. So this also breaks compatibility with old versions.

I think easiest fix is to use a regex to check if MAC and if so don't treat it as alias.

Heckie75, 2020-12-28:
Done. Pls check

JsBergbau, 2020-12-28:
Thank you very much for your fast fix. Works now.

closed 2020-12-28 by JsBergbau