Heckie75/eQ-3-radiator-thermostat #34: Window open time maximum 1 hour

JsBergbau, 2020-12-28:
Hallo Heckie75,

I've noticed a little flaw in your documentation.
```
 window <temp> <hh:mm>                          - Sets temperature in °C and period for open window mode
                                                  temp: 5.0 to 29.5 in intervals of 0.5°C, e.g. 19.5
                                                  hh:mm: time where minutes in intervals of 5 minutes, e.g. 02:05
```
Maximum window open time you can set on thermostat directly is 1:00
If you send 02:05 thermostat reports back 00:45 as window opentime. this is also shown on thermostat display.

Sorry for opening so many issues. Your project is really great and opening so many issues doesn't mean I'm picking only the non-working parts. I really like your project. 
In fact I like it so much that I've developed a Node-RED node based upon your script and so I'm testing a lot.

You can find the Node here https://flows.nodered.org/node/node-red-contrib-eq3-thermostat
I would be glad if you mention it in your Readme like:

> You can also use this code via Node-RED, see https://flows.nodered.org/node/node-red-contrib-eq3-thermostat

Heckie75, 2021-01-10:
I have pushed a fix in terms of max interval for window open mode. Pls. check it.

In terms of the requested link to node red, I would like to gather also other projects that use this script in order to give anyone the same attention.

JsBergbau, 2021-01-10:
Thanks for fixing. Checked. It works.
I've noticed this change
[image]
`-nocase` switch was introduced in commit for fixing MAC addresses in lowercase recognized as alias. 
Just checked it, script still works with lower case MACs, just wanted to assure that this was changed back by intention?

Thank you for mentioning third party projects integrating your radiator script.

Heckie75, 2021-01-10:
Yes, it was back by intention since the regex is related to aliases. When using alias "A" (capital letter) I don't want to get "Schlafzimmer" if there is also an alias for "Ankleide" (just an example in german).

JsBergbau, 2021-01-10:
Ok, thanks for explaining. I understand now.

closed 2021-01-10 by Heckie75