Heckie75/eQ-3-radiator-thermostat #36: Firmware Upgrade: Latest versions seem to require pairing

JsBergbau, 2021-01-07:
I've upgraded some "new" EQ3 which I ordered a few days ago but still had old firmware where Bluetooth was on by default and didn't report so much information on sync request like window open time.
However after that I was able anymore to control them with raspberry Pi Zero W. Even after resetting thermostat to factory default. 
After pairing them via
```
sudo bluetoothctl
scan on
<wait until mac of new Eqiva EQ3 appears>
scan off
pair <MAC>
<Enter 6 digit number, getting by longpressing control wheel>
disconnect <MAC>
```
controlling was fine again. Did a test by `remove <MAC>` in bluetoothctl and then device couldn't be controlled anymore. After repairing it could be controlled again. Important do the disconnect step in bluetoothctl otherwise eQ-3 script can't connect.

However another Rasperry PI 4 with an external bluetooth adapter can sync all 3 updated devices without beeing paired. Very strange.

Hope this information helps someone.

JsBergbau, 2021-01-08:
Ok update: With internal bluetooth of both Raspberry PI Zero W and Pi 4 I can't access Eqiva eQ3 radiator thermostat with latest firmware without pairing. Using an external bluetooth dongle however it works without pairing. Very strange.

Heckie75, 2021-01-08:
Before the update the PIN was just checked on application level (by the app based on the serial number of the device). After the update I guess that the PIN is really something that is located at bluetooth protocol. 

Unfortunately I can't test it on a Raspberry since I don't have one at this moment. I am still able to control the thermostate by using an external bluetooth adapter with my Intel NUC.

Since it seems to depend on bluetooth device I guess that it must be some kind of configuration. I remember that I had problems one or two years ago with controlling my smart meters with a Raspberry Pi Zero. The following command helped:

hciconfig hci0 sspmode disable

I think that this is some kind of configuration related to security and paring. See https://www.systutorials.com/docs/linux/man/1-hciconfig/

JsBergbau, 2021-01-08:
After the update the PIN is now 6 digits instead of 4 digits. It also changes after a factory reset, but seems to stay the same on subsquent paring button presses.
Disabling sspmode didn't help. 

A raspberry PI 3B can also control it without pairing (but on an older Raspbian release). I also think its some kind of configuration. When controlling with the App I also get questions from Android if I want to pair. I tap Cancel and still the App can control thermostat.

nrother, 2021-01-17:
I'm also having trouble controlling my valves with my RPi 3 B+ after the latest update. I tried paring them with `bluetoothctl` and turned `sspmode` off with no effect.
Strangely, it seems to work on a random basis. Retrying the same command multiple times sometime leads to success. Output when in fails, with `log_user = 1`:
```
pi@raspberrypi:~ $ ./eq3.exp 00:1A:22:0E:D7:DF manual
doCommand: received command <manual>
[                 ][LE]> connect 00:1A:22:0E:D7:DF
Attempting to connect to 00:1A:22:0E:D7:DF
Error: connect error: Function not implemented (38)
[00:1A:22:0E:D7:DF][LE]> Connection failed.
```
For me this looks like a bug in the RPI's Bluetooth Adapter (or driver)

Heckie75, 2021-02-06:
Today I have seen that one of my thermostate couldn't be controlled by the script anymore. I have updated the firmware weeks ago and everything went well until today. 

After some investigation I have paired the thermostat again like this:

ˋˋˋ
$ bluetoothctl
Agent registered
[CHG] Controller 00:1A:7D:DA:71:13 Pairable: yes
[CHG] Controller 14:F6:D8:D4:1F:F1 Pairable: yes
[bluetooth]# select  00:1A:7D:DA:71:13
Controller 00:1A:7D:DA:71:13 heckies-nuc [default]
[bluetooth]# connect 00:1A:22:0A:91:CF
Attempting to connect to 00:1A:22:0A:91:CF
[CHG] Device 00:1A:22:0A:91:CF Connected: yes
[CHG] Device 00:1A:22:0A:91:CF Connected: no
Connection successful
[CHG] Device 00:1A:22:0A:91:CF Connected: yes
Request passkey
[agent] Enter passkey (number in 0-999999): 308448
[NEW] Primary Service (Handle 0x8281)
        /org/bluez/hci1/dev_00_1A_22_0A_91_CF/service0200
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
...
[CC-RT-BLE]# disconnect
[CC-RT-BLE]# quit
ˋˋˋ

Afterwards everything works again. Don't forget to explicitly disconnect when you are in ˋˋˋbluetoothctlˋˋˋ. Otherwise the script can't connect since the thermostat is still busy.

JsBergbau, 2021-02-06:
Today I've noticed a strange error. Despite beeing paired, I couldn't control the thermostat via script any more. I've used nRFConnect on Android and didn't find the thermostat anymore. I've disabled BLE and enabled it again, afterwards I could control it again. Problem exits at least since yesterday, however today I've noticed by coincidence. 
So it might be quite a similiar issue, that an internal timer resets bluetooth or something like that.

Heckie75, 2021-10-10:
Added notes in README.md

closed 2021-10-10 by Heckie75