Heckie75/eQ-3-radiator-thermostat #39: expect: spawn id exp4 not open

gittehupe, 2021-03-05:
Hi,

the script does not really work for me. I always get the message `expect: spawn id exp4 not open`:

`./eq3.exp myEQ3 status
expect: spawn id exp4 not open
    while executing
"expect -re "\[0-9a-f\ ]*" {
    log "ok"
  }"
    (procedure "readHandle" line 10)
    invoked from within
"readHandle $HANDLES(name)"
    (procedure "readDeviceInfo" line 5)
    invoked from within
"readDeviceInfo"
    (procedure "init" line 8)
    invoked from within
"init [expr $INIT(device) + $INIT(mode) + $INIT(timer) + $INIT(serial)]"
    ("^status$" arm line 2)
    invoked from within
"switch -regexp $c {
    ^sync$ {
      init $INIT(mode)
    }
    ^comforteco.* {
      checkArgs $c 2 "comforteco" {^comforteco [0-9]{1,2}\.?[05]? [0..."
    (procedure "doCommand" line 10)
    invoked from within
"doCommand"
    (file "./eq3.exp" line 1265)
`

Any hint for solving this issue?

Thanks a lot in advance!

maciejszaferski, 2021-08-16:
Hello, I have two thermostats. First (old with oldest firmware) and second with the newest  firmware. Eq-3 with old firmware working correctly and Eq-3 with the newest firmware get the message as above. How solving this issue? Or how to downgrade firmware in Eq-3?

a-dekker, 2021-09-29:
Had the same error after updating to the latest firmware recently. First pairing with bluetoothctl as described in https://github.com/Heckie75/eQ-3-radiator-thermostat/issues/36 fixed this error for me.

closed 2021-10-10 by Heckie75